utiasASRL/pyboreas #17: Boreas-2021-09-02-11-42 ground truth poses missing/incorrect

cheneyuwu, 2022-02-28:
Two issues with the 2021/09/02 sequence:
- seems to have incorrect radar_poses.csv on AWS - need to compare AWS version with the navlab backup
- misses ground truth pose for `radar/1630598033806534.png` - maybe just remove this scan because there's another scan with timestamp 1630598033806533 (only 1ms apart)

MMOCKING, 2023-02-15:
:)  The file "radar_poses.csv" does not appear to be in the "applanix/" directory as the paper said. Have it been moved to another directory?

keenan-burnett, 2023-02-15:
@MMOCKING for which sequence, 2021-09-02 ?

MMOCKING, 2023-02-16:
> I was able to download `radar_poses.csv` for 2021-09-02. Note that some of the sequences are "test sequences" as in `pyboreas/data/splits.py`, so in order to get test results you would need to submit your odometry or localization estimates to the automated benchmark on our website. If you really need the ground truth for one of the test sequences for a paper submission, send me an email at [keenan.burnett@robotics.utias.utoronto.ca](mailto:keenan.burnett@robotics.utias.utoronto.ca) and we can discuss.

I think I downloaded the the test sequences. Thanks~

keenan-burnett, 2023-04-13:
Uploaded the ground truth radar poses stored on navlab to S3 for `boreas-2021-09-02-11-42` and deleted the duplicate radar image `radar/1630598033806534.png`

closed 2023-04-13 by keenan-burnett