utiasASRL/pyboreas #22: 3D Object Detection in Adverse Conditions?

MartinHahner, 2022-03-29:
Hi @keenan-burnett, great dataset! 🎉

Are you planning to release 3D Object Detection labels also for sequences containing rain and snowfall?
In your paper you _e.g._ mention Boreas-Objects-**V1**, so will there be a planned **V2** release?

[image]

keenan-burnett, 2022-03-29:
Hi Martin!

We're looking into this internally. Since we already have the data, it wouldn't be too much extra work to label some frames from the rain or snow sequences. I can't promise that there will be a V2 with these additions, but I'll say it could happen over the next year.

closed 2022-03-30 by MartinHahner

meiiqi, 2023-03-24:
Hi @keenan-burnett, any recent updates on this?  Thanks!

keenan-burnett, 2023-03-24:
Hi @meiiqi, we have plenty of interesting data taken during adverse weather but I currently don't have enough budget allocated for another Scale data annotation run. I know there are some researchers at my institute who are interested in pursuing this towards their winter autonomous driving research but it won't be happening any time soon.