utiasASRL/pyboreas #40: About the number of labels

ShuHengLin, 2023-07-12:
Hi, firstly, thank you for releasing this great dataset!

I noticed in your paper that you annotated a total of 7,111 lidar frames.
However, in the downloaded boreas-objects-v1/labels_detection folder, I only found 4,951 files.

I would like to inquire if any information is missing.

My download command is:
```
aws s3 sync s3://boreas/boreas-objects-v1 /path/to/boreas/boreas-objects-v1 --no-sign-reques
```

Many thanks.

keenan-burnett, 2023-07-12:
The remaining labels are held out as part of the test set.

keenan-burnett, 2023-07-12:
We have an online benchmark you can submit your results to in order to obtain results on the test set:
www.boreas.utias.utoronto.ca

ShuHengLin, 2023-07-12:
Okay, thanks for your response.