utiasASRL/pyboreas #41: Wrong tracking uuid for boreas-objects-v1

ShuHengLin, 2023-07-14:
Hi, I noticed that in the labels of boreas-objects-v1, there are several frames where the tracking uuid is inconsistent with the previous frame (in the same scene).
This causes an issue when visualize these frames. The function **bbs.interpolate()** treats all the bounding boxes as "non-overlapping boxes" and deletes them, resulting in no boxes to being drawn.

* I think the following files have inconsistent tracking uuids:
```
1598987807960198.txt → 1598987808167660.txt
1598987873812603.txt → 1598987874019891.txt
1598988085990765.txt → 1598988086198228.txt
1598988105902338.txt → 1598988106109670.txt
1598988218835735.txt → 1598988219043182.txt
1598988238954403.txt → 1598988239161828.txt
1598990011979599.txt → 1598990012187036.txt
1598990969792240.txt → 1598990969999645.txt
```

* The visualization code:
```Python
import matplotlib.pyplot as plt
import numpy as np

from pyboreas import BoreasDataset

bd = BoreasDataset(root='/data_1TB_2/boreas/', split=None, verbose=False, labelFolder="labels_detection")
seq = bd.sequences[0]
seq.filter_frames_gt()
seq.synchronize_frames('lidar')

index = [[930, 931],
         [1122],
         [1314, 1315],
         [1410, 1411],
         [1601],
         [1699],
         [3714, 3715],
         [4278, 4279],
        ]

for i in range(len(index)):
  for j in index[i]:
    cam = seq.get_camera(j)
    bbs = cam.get_bounding_boxes(seq.labelFiles, seq.labelTimes, seq.labelPoses)
    bbs.visualize(cam.img, seq.calib.P0, checkdims=True)
    cam.visualize()
    cam.unload_data()
```

keenan-burnett, 2023-07-17:
Hmm, It's possible that this is just how our labels were produced.
For now, could you consider these to be separate scenes?
For example, (1598987807960198.txt → 1598987808167660.txt) could be considered the boundary between two scenes.
It's possible that I incorrectly assumed that the uuids would be consistent between these scenes when the original labelling project did not achieve this. I think that's the most likely case.

I'll have to increase the number of scenes accordingly and split them up using similar timestamps to what you have provided.
I don't have time to look into it right now but I'll add it to my dataset-related TODO list.

ShuHengLin, 2023-07-18:
Got it, thank you for your explanation.

closed 2024-11-14 by keenan-burnett