utiasASRL/pyboreas #42: Raw IMU data

Nader-13, 2023-11-22:
Hello,

The docs mention that raw imu data are provided as (imu_raw.csv) which can be extracted from raw Applanix logs. 
The log files don't seem to be readable and I couldn't find a standalone csv file for imu_raw.
Can you please give a bit more details on this?

Thanks.

keenan-burnett, 2023-11-22:
Which sequence are you looking at?

The raw IMU data is available at `<sequence>/applanix/imu_raw.csv`

keenan-burnett, 2023-11-22:
The raw IMU data is not available for `boreas-objects-v1` but should be present in the other sequences.

Nader-13, 2023-11-22:
I checked:

boreas-2020-12-01-13-26
boreas-2021-01-26-11-22
boreas-2021-04-08-12-44
boreas-2021-04-22-15-00
boreas-2021-04-29-15-55
boreas-2021-06-17-17-52
boreas-2021-07-20-17-33

All are part of the odometry and/or localization splits according to [splits.py](https://github.com/utiasASRL/pyboreas/blob/master/pyboreas/data/splits.py), but all I could find is the post-processed imu.csv data.

Appreciate your help,

keenan-burnett, 2023-11-22:
The `imu_raw.csv` file is present in those sequences, how are you trying to download this?

Nader-13, 2023-11-22:
I use the bash scripts generated by the official website.
I just re-downloaded the applanix folder for those sequences and, to my surprise, all the imu_raw.csv files are there.
My 1st download was a long time ago, not sure how these files were missed at that time.  Sorry for the false alarm!

Thanks a lot for your help, I will close the issue.

closed 2023-11-22 by Nader-13