utiasASRL/pyboreas #44: Question about IMU and Wheel data

Tompson11, 2024-02-29:
Hi! Thanks for sharing your dataset and I have some questions about it:

1. Are data of `imu.csv` and `imu_raw.csv` in the same coordinate frame (i.e. IMU body frame) ?
2. Are IMU body frame and Applanix reference frame the same one ? If so, I notice that in `DATA_REFERENCE.md`

> IMU body frame which is defined as x-backwards, y-left, z-up

while the figure of sensor placement shows that Applanix reference frame is defined as x-right, y-forwards, z-up. Could you explain the difference?
3. `dmi.csv` provides wheel encoder data and I'd like to use them for odometry. Could you please share more information about it such as:

- On which wheel the dmi sensor is mounted?
-  how many pulse counts correspond to a whole circle?

keenan-burnett, 2024-02-29:
imu.csv is in the applanix frame, x-right, y-forwards, z-up, this is the post-processed output of the applanix solution.

imu_raw.csv is x-backwards, y-left, z-up.

Imu body frame and applanix frame are not the same thing.

The wheel encoder is mounted on the back left wheel.

The DMI scale factor is roughly 1916 pulses per metre but will likely vary from sequence to sequence.

Tompson11, 2024-02-29:
Thank you so much!

closed 2024-04-18 by keenan-burnett

paaraujo, 2024-06-26:
Hi @keenan-burnett , just wanted to check if

imu_raw.csv is x-backwards, y-left, z-up

is correct. I checked the csv file and it seems that the IMU followed the right-hand approach. Thus, z is down instead of up. Is my interpretation correct?

keenan-burnett, 2024-06-26:
Sorry, that should have been x-backwards, y-left, z-down, you're correct.