utiasASRL/pyboreas #49: A runtime error in executing readme script

behroozbc, 2024-05-30:
I wanted to run the readme script but I got a runtime error. 
This is error
```
Exception has occurred: RuntimeError       (note: full exception trace is shown but execution is paused at: <module>)

        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.

        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:

            if __name__ == '__main__':
                freeze_support()
                ...

        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.
  File "E:\Work\University\sensor\win\genrate-boreas-image\main.py", line 5, in <module>
    bd = BoreasDataset(root)
  File "<string>", line 1, in <module> (Current frame)
RuntimeError: 
        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.

        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:

            if __name__ == '__main__':
                freeze_support()
                ...

        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.
```
the full script 
```
import numpy as np
from pyboreas import BoreasDataset

root = 'J:/boreas/'
bd = BoreasDataset(root)

# Note: The Boreas dataset differs from others (KITTI) in that camera,
# lidar, and radar measurements are not synchronous. However, each
# sensor message has an accurate timestamp and pose instead.
# See our tutorials for how to work with multiple sensors.

# Loop through each frame in order (odometry)
for seq in bd.sequences:
    # Iterator examples:
    for camera_frame in seq.camera:
        img = camera_frame.img  # np.ndarray
        # do something
        camera_frame.unload_data() # Memory reqs will keep increasing without this
    for lidar_frame in seq.lidar:
        pts = lidar_frame.points  # np.ndarray (x,y,z,i,r,t)
        # do something
        lidar_frame.unload_data() # Memory reqs will keep increasing without this
    # Retrieve frames based on their index:
    N = len(seq.radar_frames)
    for i in range(N):
        radar_frame = seq.get_radar(i)
        # do something
        radar_frame.unload_data() # Memory reqs will keep increasing without this

# Iterator example:
cam_iter = bd.sequences[0].get_camera_iter()
cam0 = next(cam_iter)  # First camera frame
cam1 = next(cam_iter)  # Second camera frame

# Randomly access frames (deep learning, localization):
N = len(bd.lidar_frames)
indices = np.random.permutation(N)
for idx in indices:
    lidar_frame = bd.get_lidar(idx)
    # do something
    lidar_frame.unload_data() # Memory reqs will keep increasing without this

# Each sequence contains a calibration object:
calib = bd.sequences[0].calib
point_lidar = np.array([1, 0, 0, 1]).reshape(4, 1)
point_camera = np.matmul(calib.T_camera_lidar, point_lidar)

# Each sensor frame has a timestamp, groundtruth pose
# (4x4 homogeneous transform) wrt a global coordinate frame (ENU),
# and groundtruth velocity information. Unless it's a part of the test set,
# in that case, ground truth poses will be missing. However we still provide IMU
# data (in the applanix frame) through the imu.csv files.
lidar_frame = bd.get_lidar(0)
t = lidar_frame.timestamp  # timestamp in seconds
T_enu_lidar = lidar_frame.pose  # 4x4 homogenous transform [R t; 0 0 0 1]
vbar = lidar_frame.velocity  # 6x1 vel in ENU frame [v_se_in_e; w_se_in_e]
varpi = lidar_frame.body_rate  # 6x1 vel in sensor frame [v_se_in_s; w_se_in_s]
```
I have two boreas datasets with all details ( exclude nothing)

keenan-burnett, 2024-05-30:
I wasn't able to reproduce. This seems like an issue related to using `multiprocessing` on Windows:
https://stackoverflow.com/questions/18204782/runtimeerror-on-windows-trying-python-multiprocessing

Have a look at that stack overflow to see if it resolves your issue.

closed 2024-06-04 by keenan-burnett

behroozbc, 2024-06-21:
I run the script on my linux system and it give me the result.