utiasASRL/pyboreas #50: IndexError: list index out of range for synchronize_frames

behroozbc, 2024-06-21:
I want to synchronize frames with ref to camera, but I got this error message
```
list index out of range
  File "/home/behro/work/uni/convertfiles/broeas-prepare-script/main.py", line 15, in <module>
    seq.synchronize_frames(ref='camera')
IndexError: list index out of range
```
the script 
```
bd = BoreasDataset(pathtodset, verbose=True)
for seq in bd.sequences:
    seq.synchronize_frames(ref='camera')
```
This is the console message at loading datasets.
```
WARNING: this may be a test sequence, or the groundtruth is not yet available: boreas-2021-01-26-10-59
WARNING: this may be a test sequence, or the groundtruth is not yet available: boreas-2020-12-04-14-00
WARNING: this may be a test sequence, or the groundtruth is not yet available: boreas-2021-10-05-15-35
SEQ: boreas-2020-12-04-14-00
camera frames: 24510
lidar frames: 11923
radar frames: 0
-------------------------------
SEQ: boreas-2021-01-26-10-59
camera frames: 12308
lidar frames: 11867
radar frames: 0
-------------------------------
SEQ: boreas-2021-04-29-15-55
camera frames: 10719
lidar frames: 10336
radar frames: 4287
-------------------------------
SEQ: boreas-2021-06-17-17-52
camera frames: 12349
lidar frames: 11906
radar frames: 4939
-------------------------------
SEQ: boreas-2021-07-27-14-43
camera frames: 10013
lidar frames: 9657
radar frames: 4006
-------------------------------
SEQ: boreas-2021-10-05-15-35
camera frames: 12503
lidar frames: 12074
radar frames: 0
-------------------------------
SEQ: boreas-2021-10-22-11-36
camera frames: 10104
lidar frames: 9733
radar frames: 4042
-------------------------------
total camera frames: 92506
total lidar frames: 77496
total radar frames: 17274
```

I think this error is related to the package not my script, because my script does not do anything at that moment!

keenan-burnett, 2024-06-22:
I need more information to help

keenan-burnett, 2024-06-22:
I noticed you're missing radar data for the two sequences that have errors, does it work for the sequences with radar data?

I need to see the folder structure for the sequences that are giving you errors

behroozbc, 2024-06-25:
I have problem with this datasets `boreas-2020-12-04-14-00` ,`boreas-2021-01-26-10-59` and  `boreas-2021-10-05-15-35`.
Now I see I think I can be related to radar dataset, I am going to download completing my datasets and send the result.
my folder structure is the same as downloading with aws and I did not change anything. 
I run the command like this
```
$root=.
aws s3 sync s3://boreas/boreas-2021-04-29-15-55  $root/boreas-2021-04-29-15-55 --exclude "*"  --include "lidar/*" --include "camera/*" --include "applanix/*" --include "calib/*"
```
I got the command from the website download section!

behroozbc, 2024-06-26:
This is result of `tree -d`
```
.
├── boreas-2020-12-04-14-00
│   ├── applanix
│   ├── calib
│   ├── camera
│   ├── lidar
│   └── radar
│       ├── cart
│       └── mask
├── boreas-2021-01-26-10-59
│   ├── applanix
│   ├── calib
│   ├── camera
│   ├── lidar
│   └── radar
│       ├── cart
│       └── mask
├── boreas-2021-04-29-15-55
│   ├── applanix
│   │   ├── applanix
│   │   └── raw_logs
│   ├── calib
│   ├── camera
│   ├── lidar
│   └── radar
│       ├── cart
│       └── mask
├── boreas-2021-06-17-17-52
│   ├── applanix
│   │   └── raw_logs
│   ├── calib
│   ├── camera
│   ├── lidar
│   └── radar
│       ├── cart
│       └── mask
├── boreas-2021-07-27-14-43
│   ├── applanix
│   │   └── raw_logs
│   ├── calib
│   ├── camera
│   ├── lidar
│   └── radar
│       ├── cart
│       └── mask
├── boreas-2021-10-05-15-35
│   ├── applanix
│   ├── calib
│   ├── camera
│   ├── lidar
│   └── radar
│       ├── cart
│       └── mask
└── boreas-2021-10-22-11-36
    ├── applanix
    │   └── raw_logs
    ├── calib
    ├── camera
    ├── lidar
    └── radar
```

keenan-burnett, 2024-06-26:
Can you provide the full stack trace of your error?

behroozbc, 2024-06-26:
I downloaded the radar dataset and the error is not showing up. It definitely related to radar dataset.

keenan-burnett, 2024-06-26:
okay, so we can close this issue as resolved?

behroozbc, 2024-06-26:
I found the problem, but I think it is good to fix this issue on code.

closed 2024-06-26 by keenan-burnett