utiasASRL/pyboreas #51: Data missing

paaraujo, 2024-06-26:
Hi,

I downloaded a few trajectories (~30) and I'd like to report that for the following trajectories the folder `applanix` only contains the files `dmi.csv` and `imu_raw.csv`.

- boreas-2021-02-09-12-55
- boreas-2021-03-09-14-23
- boreas-2021-04-22-15-00
- boreas-2021-06-29-20-43
- boreas-2021-09-08-21-00
- boreas-2021-09-09-15-28
- boreas-2021-10-05-15-35
- boreas-2021-11-06-18-55

Best, Paulo.

keenan-burnett, 2024-06-26:
This is to be expected, these sequences are a part of the test set:
https://github.com/utiasASRL/pyboreas/blob/eeb2b1bb302f5386eff7198858846f6d89675fda/pyboreas/data/splits.py#L42-L56

paaraujo, 2024-06-26:
Nice! Good to know, I am using the dataset independent of the splits provided. Thank you!

closed 2024-06-26 by paaraujo