utiasASRL/pyboreas #52: lid.project_onto_image return zero arrays for uv and colors.

behroozbc, 2024-06-26:
I tested this code in two part of the datasets `boreas-2021-01-26-10-59` and `boreas-2021-10-05-15-35`, which return a zero arrays for uv and colors. 

https://github.com/utiasASRL/pyboreas/blob/eeb2b1bb302f5386eff7198858846f6d89675fda/pyboreas/tutorials/lidar_camera_projection.ipynb#L114

my code 
```
for idx in range(len(seq.lidar_frames)):
            try:
                cam = seq.get_camera(idx)
                lid = seq.get_lidar(idx)
                lid.remove_motion(lid.body_rate)
                T_enu_camera = cam.pose
                T_enu_lidar = lid.pose
                T_camera_lidar = np.matmul(get_inverse_tf(T_enu_camera), T_enu_lidar)
                lid.transform(T_camera_lidar)
                lid.passthrough([-75, 75, -20, 10, 0, 40])  # xmin, xmax, ymin, ymax, zmin, zmax
                # Project lidar points onto the camera image, using the projection matrix, P0.
                uv, colors, _ = lid.project_onto_image(seq.calib.P0)
                fig = plt.figure(figsize=(24.48, 20.48), dpi=100)
                ax = fig.add_subplot()
                ax.imshow(np.zeros(cam.img.shape, dtype=np.uint8)) # you can change it to this `np.zeros(cam.img.shape, dtype=np.uint8)` for black background
                ax.set_xlim(0, 2448)
                ax.set_ylim(2048, 0)
                ax.scatter(uv[:, 0], uv[:, 1], c=colors, marker=',', s=3, edgecolors='none', alpha=0.7, cmap='jet')
                ax.set_axis_off()

```
I downloaded the full dataset for this two parts. 
If any more information is require, please tell me.

behroozbc, 2024-06-28:
I found the same problem in `boreas-2021-11-28-09-18` part too

keenan-burnett, 2024-06-28:
Hey, what you're trying to do will not work for those sequences because those sequences are in the test set, where no ground truth poses are provided. So, when you do `cam.pose` or `lid.pose`, they're just returning an identity matrix.

You can try this again with a sequence that is a part of the training set. You can find out which sequences are a part of the training / test splits here:

https://github.com/utiasASRL/pyboreas/blob/eeb2b1bb302f5386eff7198858846f6d89675fda/pyboreas/data/splits.py#L42-L56

closed 2024-06-28 by keenan-burnett