utiasASRL/pyboreas #55: intensity vs calibrated reflectivity

ed-cho, 2024-10-30:
Hi, thanks for the great dataset. However, I've noticed that the [guide](https://github.com/utiasASRL/pyboreas/blob/master/DATA_REFERENCE.md#lidar) describes the LiDAR point clouds to be in the format [x, y, z, i, r, t] where i is the intensity of the reflected infrared signal. However, the [user manual](https://data.ouster.io/downloads/velodyne/user-manual/alpha-prime-rev4-user-manual.pdf) of the Velodyne Alpha Prime (VLS-128) states that it reports calibrated reflectivity, not intensity. Can I assume that the i value is actually calibrated reflectivity? Or did you use another approach to convert the calibrated reflectivity back to intensity?

keenan-burnett, 2024-11-01:
Hmm, I think you're correct that these values are reflectivity.
This is the line from our sensor driver that works with the raw velodyne data:
https://github.com/utiasASRL/velodyne_vls/blob/7ad7ff3b609a4bddab574f41ea1eb41852efb783/velodyne_pointcloud/src/lib/rawdata.cc#L890

It's a bit confusing but the driver refers to this as intensity, probably for legacy reasons when working with older Velodyne models.

ed-cho, 2024-11-01:
Makes sense. Thanks for confirming!

closed 2024-11-05 by keenan-burnett