utiasASRL/pyboreas #57: Something about benchmarks

yangsc4063, 2024-11-14:
If I want to do pure radar object detection, the test results will be in the radar coordinate system and published with radar timestamps. How can I test it on the benchmark? 
Thank you for your work, looking forward to your reply.

keenan-burnett, 2024-11-14:
Can you try transforming your detections into the nearest lidar frame and evaluating that way?

keenan-burnett, 2024-11-14:
Try running `pyboreas/eval/detection.py` on a validation set locally with `--radar True` and see what results you get. The website benchmark uses the same format as this script.

yangsc4063, 2024-11-15:
> Try running `pyboreas/eval/detection.py` on a validation set locally with `--radar True` and see what results you get. The website benchmark uses the same format as this script.

Thank you for your reply. I am just planning to do such a job now, but it has not been realized yet. I will try it later. Thank you again!

closed 2024-12-03 by yangsc4063