utiasASRL/pyboreas #63: About rotation_z angle of labels

leehungu, 2025-04-10:
Hi,
Thank you for providing the awesome dataset. I run the script of 'pyboreas/pyboreas/tutorials/object_detection.ipynb',  set the 'seq_ind=0' and 'index=0', subsequently get the below output. I'm not sure is there something wrong?

[image]

Petros626, 2025-06-30:
@leehungu in camera coordinates a 3d bbox has rotation_y (bc of camera coordinates). When you tell me in which format and frame the groundtruth labels are provided, I could help you.

leehungu, 2025-07-06:
> This looks like a visualization issue rather than an issue with the bounding box labels since the rectangle is oriented correctly here with the long side of the rectangular prism lined up with the orientation of the truck?

I'm not sure where the issue lies, but it seems to be caused by a 90-degree difference in the true values, or it could be a visualization error. I ran the tutorials ('pyboreas/pyboreas/tutorials/object_detection.ipynb') without making any changes except updating the data path.

leehungu, 2025-07-06:
> So from my experience the annotations has to be transformed to the camera coordinate system with the right box corner points format. For me it looks like the camera 3d box points aren't correct.

Thanks a lot. If I recall correctly, I have attempted to convert the boxes into the camera coordinate system; however, they do not appear to be visualized correctly.

Petros626, 2025-07-06:
The reference point of a lidar box is in the center, for a camera box it's on ground. So you have to use calib Infos for the frame transformation and set correctly the box itself e.g. z =height/2. These information count for KITTI dataset but maybe here's the same case.

Petros626, 2025-07-06:
It was just for you to do a code checkup for the transformation process, you know? As I said the order of the box corner points could be the issue. I don't know youre code, but maybe this helps:
```txt
    # https://github.com/kuixu/kitti_object_vis/blob/12ce0a2348f6e1405c502bf32e51d76d3a970396/kitti_util.py#L669
      1 -------- 0
     /|         /|
    2 -------- 3 .
    | |        | |
    . 5 -------- 4
    |/         |/
    6 -------- 7
```
My transformation from KITTI LiDAR 3D BBox to Camera 3D BBox.
[image]

leehungu, 2025-07-06:
> It was just for you to do a code checkup for the transformation process, you know? As I said the order of the box corner points could be the issue. I don't know youre code, but maybe this helps:
> 
>     # https://github.com/kuixu/kitti_object_vis/blob/12ce0a2348f6e1405c502bf32e51d76d3a970396/kitti_util.py#L669
>       1 -------- 0
>      /|         /|
>     2 -------- 3 .
>     | |        | |
>     . 5 -------- 4
>     |/         |/
>     6 -------- 7
> My transformation from KITTI LiDAR 3D BBox to Camera 3D BBox: <img alt="Image" width="1013" height="367" src="https://private-user-images.githubusercontent.com/62354721/462926171-ebdc51da-0eb5-4b29-bf2a-1fb30ba601ea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTE4MTE5NTksIm5iZiI6MTc1MTgxMTY1OSwicGF0aCI6Ii82MjM1NDcyMS80NjI5MjYxNzEtZWJkYzUxZGEtMGViNS00YjI5LWJmMmEtMWZiMzBiYTYwMWVhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzA2VDE0MjA1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZkNjk4M2UwYmZmMDc2YjlhNjVmYWFhMTViMTczMDQzYWQ3NmE3ZTE4NzBhOTRmNmZkZTEyMDYwYzVkMDU1MmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.wFMNAVz-AGU0jBoyMoO4g2i9Pd59IiOuB28y-CX9MNE">

Sorry, I do not get the point. But, it is true there is a difference about yaw  between lidar and camera system.  And  I ran the tutorials of this code ('pyboreas/pyboreas/tutorials/object_detection.ipynb') without making any changes except updating the data path. You can find the all code in this repo. Anyway, thanks.

Petros626, 2025-07-06:
The yaw isn't your problem. It's how the box gets created.